ARM-software/SCALE-Sim #7: Ofmap write to DRAM after the execution of each layer

menglioculus, 2018-10-24:
In the current code, fill buffer and drain buffer are always drained after the execution of each layer, which means the ofmaps are always written back to the DRAM. Is this necessary?

AnandS09, 2018-10-30:
Hi @menglioculus, you are right that there can be designs where the outputs can be kept in the accelerator and reused for computations in the next layer. However, in current CNNs, the convolution operation is always followed by some activation or pooling operation. In SCALE-Sim we do not model the functional units for doing these operations hence the tool always drains out the data.

HaFred, 2021-04-10:
So why is the DRAM read & write behavior is different in terms of the usage of sram buffer? Did you solve this issue? @menglioculus Thanks.
https://github.com/ARM-software/SCALE-Sim/blob/5825f6cee26bac223ce6019fcfc4f384c1142cb6/dram_trace.py#L30
https://github.com/ARM-software/SCALE-Sim/blob/5825f6cee26bac223ce6019fcfc4f384c1142cb6/dram_trace.py#L116

AnandS09, 2021-04-23:
Hi @HaFred , 

Not sure if I understand the issue you mentioned correctly. Can you please elaborate? 
Plus, it might be a good idea to create a new issue to track this. We can close this then.

@whatmough : Paul let's close this one. 

Thanks
Anand

HaFred, 2021-04-25:
> Hi @HaFred ,
> 
> Not sure if I understand the issue you mentioned correctly. Can you please elaborate?
> Plus, it might be a good idea to create a new issue to track this. We can close this then.
> 
> @whatmough : Paul let's close this one.
> 
> Thanks
> Anand

Hi Anand,

Like I mentioned above, I did not get when we are doing modeling, why is the `dram_trace_read_v2` function model the SRAM as a single set, while the `dram_trace_write` function model the SRAM as two-sets of double buffer?

Thank you.
Fred

closed 2023-05-11 by danh-arm