ARM-software/SCALE-Sim #14: TypeError: range() integer end argument expected, got float

tzhang0924, 2018-12-11:
$ python scale.py
('Using Architechture from ', './configs/scale.cfg')
====================================================
******************* SCALE SIM **********************
====================================================
Array Size: 	32x32
SRAM IFMAP: 	512
SRAM Filter: 	512
SRAM OFMAP: 	256
CSV file path: 	./topologies/alexnet.csv
Dataflow: 	Output Stationary
====================================================

Commencing run for Conv1
Generating traces and bw numbers
100%|██████████████████████████████████████████████████████████████████████████████████| 8748/8748 [00:06<00:00, 1290.77it/s]
Traceback (most recent call last):
  File "scale.py", line 201, in <module>
    app.run(main)
  File "/Library/Python/2.7/site-packages/absl/app.py", line 300, in run
    _run_main(main, args)
  File "/Library/Python/2.7/site-packages/absl/app.py", line 251, in _run_main
    sys.exit(main(argv))
  File "scale.py", line 198, in main
    s.run_scale()
  File "scale.py", line 93, in run_scale
    self.run_once()
  File "scale.py", line 129, in run_once
    offset_list = offset_list
  File "/Users/t.zhang/work/SCALE-Sim/run_nets.py", line 108, in run_net
    dram_ofmap_trace_file= net_name + "_" + name + "_dram_ofmap_write.csv"
  File "/Users/t.zhang/work/SCALE-Sim/trace_gen_wrapper.py", line 80, in gen_all_traces
    dram_trace_file=dram_ifmap_trace_file,
  File "/Users/t.zhang/work/SCALE-Sim/dram_trace.py", line 91, in dram_trace_read_v2
    for _ in range(words_per_cycle):
TypeError: range() integer end argument expected, got float.

tzhang0924, 2018-12-11:
add int() to force the words_per_cycle to be integer solve the problem. But I am not sure if anyone else sees the same issue. If so, I can commit the fix. 

diff --git a/dram_trace.py b/dram_trace.py
index 85e4a0b..347c4de 100644
--- a/dram_trace.py
+++ b/dram_trace.py
@@ -81,7 +81,7 @@ def dram_trace_read_v2(

         # Generate the filling trace from time t_fill_start to t_drain_start
         cycles_needed = t_drain_start - t_fill_start
-        words_per_cycle = math.ceil(len(sram) / (cycles_needed * word_sz_bytes))
+        words_per_cycle = int(math.ceil(len(sram) / (cycles_needed * word_sz_bytes)))

tzhang0924, 2018-12-11:
similar issues popped up. The following change gives the full fix. 

diff --git a/dram_trace.py b/dram_trace.py
index 85e4a0b..998a2c5 100644
--- a/dram_trace.py
+++ b/dram_trace.py
@@ -52,7 +52,7 @@ def dram_trace_read_v2(

                     # Generate the filling trace from time t_fill_start to t_drain_start
                     cycles_needed   = t_drain_start - t_fill_start
-                    words_per_cycle = math.ceil(len(sram) / (cycles_needed * word_sz_bytes))
+                    words_per_cycle = int(math.ceil(len(sram) / (cycles_needed * word_sz_bytes)))

                     c = t_fill_start

@@ -81,7 +81,7 @@ def dram_trace_read_v2(

         # Generate the filling trace from time t_fill_start to t_drain_start
         cycles_needed = t_drain_start - t_fill_start
-        words_per_cycle = math.ceil(len(sram) / (cycles_needed * word_sz_bytes))
+        words_per_cycle = int(math.ceil(len(sram) / (cycles_needed * word_sz_bytes)))

         c = t_fill_start

@@ -137,7 +137,7 @@ def dram_trace_write(ofmap_sram_size = 64,
             #print("Draining data. CLK = " + str(clk))
             if len(sram_buffer[draining_buf]) > 0:
                 delta_clks = clk - last_clk
-                data_per_clk = math.ceil(len(sram_buffer[draining_buf]) / delta_clks)
+                data_per_clk = int(math.ceil(len(sram_buffer[draining_buf]) / delta_clks))
                 #print("Data per clk = " + str(data_per_clk))

AnandS09, 2018-12-31:
These issues are popping up due to python2.7. Python3 works fine. 
I can see that we missed this in the readme which is causing much confusion. 
Thank you for pointing this out, see if you still get the same issues with python3.

tzhang0924, 2019-01-02:
@AnandS09 , changing to python3 eliminates the issue. It would be great if you can update the README to point it out. Thanks.

closed 2019-01-02 by tzhang0924

whatmough, 2019-01-02:
Thanks Tao!!

markfilan, 2022-01-12:
The problem is mainly because of the float value in range function. [Python range](http://net-informations.com/python/pro/xrange.htm) function does not accept floating point numbers as arguments, so you must use integers.

Range(start value, end value, step)

If it's old code, it would have worked on Python 2. Integer division in Python 2 produces another integer. In Python 3 it produces a float.