ARM-software/SCALE-Sim #15: Estimate cycles

hex0102, 2019-01-06:
Hi,
It seems like the cycle estimated are just for computation.
When the data is not in SRAM, the cycles from loading the data from DRAM are not included in the cycles for this layer.
Please correct me if I'm wrong.
Thanks.

AnandS09, 2019-01-07:
Yes, they are not included in the compute cycle count. The rationale behind this decision is that the compute in scale-sim is assumed to start once the memory buffer is filled. 
However, scale-sim does capture the data movement time and you can see this in two places:
1) The <topology>_details.csv file in the outputs folder.
2) In the DRAM trace files for layers, you will notice that some data is moved in -ve cycles. These traces pertain to data moved before the compute starts.

Let me know if this helps.

hex0102, 2019-01-10:
>    1. The _details.csv file in the outputs folder.

Thank you for the reply. Once the computation starts, the double buffer
design hide the dram latency. right?

I also have another question about the input buffer design. How is the
input buffer designed and the data organized (or reorganized) so that the
data fetched per cycle is enough to feed the systolic array and also in a
correct order?

Thank you.

AnandS09, 2019-01-16:
Please find the responses below.

>Once the computation starts, the double buffer design hide the dram latency. right?
Yes, that is correct. You will notice in SCALE-Sim give you bandwidth requirements, which the system needs to support to ensure that the double buffering happens without any glitches and DRAM latency is always hidden.

>How is the input buffer designed and the data organized (or reorganized) so that the data fetched per cycle is enough to feed the systolic array and also in the correct order?
SCALE-Sim does not model any implementation specific details. Since this is not a functional simulator, we assume that the memory buffer has mechanisms to supply the required data every cycle. 
Actual implementations might achieve this in multiple ways such as (i) having an operand collector and multi banked SRAM as in the case of GPUs, or (ii) having a clever interconnect etc.

hex0102, 2019-01-16:
Thank you very much for the clarification.
This helps a lot.

closed 2019-01-16 by hex0102