ARM-software/SCALE-Sim #18: Read partialsums from Output SRAM

hex0102, 2019-01-18:
Hi, @AnandS09 
Thank you for the excellent simulator.  In the modeling of weight stationary dataflow, if the height of the systolic array is smaller the number of Conv_window, the process has to be folded (which is very well modeled in the simulator). But the partial sum from one fold has to be fetched first, then added to another fold, and finally, the accumulation results are formed.
It seems like fetching the partial sum is not simulated in the accelerator. I'm not sure if I missed some critical code. Could you clarify a little for me?
Thank you very much.

hex0102, 2019-01-18:
@AnandS09 
Thank you for the clarification. Your answer perfectly addressed my questions.
Btw, do you have any plans to further improve the simulator later? Like the partial-sum I mentioned, multi-level memory or NOCs.

whatmough, 2019-01-30:
Yes!  We plan to continue developing the simulator, as it's been very
useful for our various projects in this area.  We'd really welcome
contributors also...

AnandS09, 2019-02-21:
Hi @hex0102 ,

As @whatmough mentioned, we are working to add more features like the ones you mentioned. However, if you have made some changes and think it would be a good addition, please feel free to send a pull request.

whatmough, 2021-04-30:
We will work on this in version 2, which can be found here: https://github.com/scalesim-project/scale-sim-v2

closed 2021-04-30 by whatmough