ARM-software/SCALE-Sim #19: Obtaining area and power

pakiko24, 2019-02-05:
Hello, 

I execute SCALE-SIM with MobileNets and the tool generates the corresponding cycle-accurate accesses to DRAM memory (and SRAM). The traces are like this:

```
-87, 10000384.0, 10000385.0, 10000386.0, 10000387.0, 10000388.0, 10000389.0, 10000390.0, 10000391.0, 10000392.0, 10000393.0, 
-86, 10000394.0, 10000395.0, 10000396.0, 10000397.0, 10000398.0, 10000399.0, 10000400.0, 10000401.0, 10000402.0, 10000403.0, 
-85, 10000404.0, 10000405.0, 10000406.0, 10000407.0, 10000408.0, 10000409.0, 10000410.0, 
```

What does this data mean? reading the paper I can suppose that the first number is the cycle and the others in the same row are the memory addresses that are accessed in that cycle. However, in that case, why are the addresses in floating point representation? and why is the cycle  negative?

In the case this is not the representation, what do all of this mean? How can I estimate power and area using this traces? I have been looking up DRAMPower but it seems the trace formats are different. Do I have to modify the traces in order fit the ones accepted by DRAMPower?

Thank you very much,
Francisco.

AnandS09, 2019-02-21:
Hi @pakiko24 ,

SCALE-Sim's traces follow the following format,
`cycle_num, <address1, address2,...., addressN>`

I think the conversion to DRAMPower's trace format is not too involved from this format. 
@yuhao : Probably you can provide some more insights.

pakiko24, 2019-02-26:
Hi @AnandS09 ,

Sure, I think it is not difficult to turn the traces into the DRAMPower format. Anyway, could I know exactly how you get the energy and power numbers of your paper?

AnandS09, 2019-02-26:
Hi @pakiko24, 

For one of our papers, my colleague @yuhao used a script to convert SCALE-Sim traces to DRAMPower. For the arXiv paper, we used got energy per SRAM read/write, DRAM read/write and MAC operation from synthesis. We then used the SCALE-sim traces to calculate energy consumption for the entire operation. Power numbers were then obtained for various frequencies.

whatmough, 2019-02-26:
It would be good to have this conversion script in the scale sim
repository.  I think it will be very helpful for others also...

pakiko24, 2019-02-27:
Thank you for your quick answer @AnandS09 ! It is really useful for me. Are you thinking of realising  the scripts soon as @whatmough say? If not, please let me know it and I will code one.

Btw, thank you very much for your excellent simulator.  I am using it and really helps me!

whatmough, 2019-05-18:
It would be great if you could code one up!!  :)

wxbbuaa2011, 2019-06-26:
we can obtain the area and energy from the dram and sram traces generated by SCALE-Sim. That's very good.

wxbbuaa2011, 2019-06-26:
@ whatmough @pakiko24 @AnandS09 How can I obtain the area and energy from the dram and sram traces generated by SCALE-Sim? I also want to konw how to obtain the area and energy. Thank you very much.

wxbbuaa2011, 2019-06-26:
@ whatmough @pakiko24 How can I obtain the area and energy from the dram and sram traces generated by SCALE-Sim? I also want to konw how to obtain the area and energy. Thank you very much.

pakiko24, 2019-06-26:
Dear @wxbbuaa2011 ,

you can get area and power by turning the traces into DRAMSimv2 format. You could code a script with that aim.

whatmough, 2021-04-30:
The power/performance/area modeling aspect is somewhat out of scope for the simulator at the current time.  However, we are open to contributions in this area.

closed 2021-04-30 by whatmough