ARM-software/SCALE-Sim #21: Simulation stalls after running Conv2 of AlexNet

lehahoang, 2019-02-25:
Hi,
First and foremost, I would like to say thank you for releasing the open-source simulator.
I have an issue with running scale.py. It stalls after generating traces and bw numbers of Conv2 as you can see from the screenshot. 

[image]
I am using Python 3.7.1.
Thank you.

pakiko24, 2019-02-26:
Hi @lehahoang,
Have you tried with the version of Python 3.5? I think that issue could be the problem. For instance, I have executed the simulator using version 2.7 and it does not work well. Indeed I've had  to modify the simulator in order to adapt the simulator to this version. It seems the simulator is not easily adaptable to other versions of Python.

AnandS09, 2019-02-26:
Hi @lehahoang , @pakiko24 ,

The simulator should work for any python3 version. I saw @lehahoang 's log and noticed that Alexnet conv2 took 38mins to run (maybe a machine issue). My thought is that scale-sim is still running in the background, parsing the SRAM traces to generate DRAM bandwidth requirements and additional traces. 
@lehahoang Can you please give a run overnight and see how much time it takes to complete. Also, if possible share the configuration of the machine you are running the tool on.

lehahoang, 2019-02-26:
Hi @pakiko2, @AnandS09 
I have never run the simulator with Python 3.5. I suppose the simulator had no problem with Python 3.7.1; the issue came from my PC.  Now everything is fine with the simulator. Thank you for your considerations.
@AnandS09 Can you please put the correct path of the default network in scale.py file. It should not be ./topologies/alexnet.csv  up to the moment that I write this comment. Thank you.

lehahoang, 2019-02-26:
Thanks @AnandS09 for the support. I think this issue should now be closed.

closed 2019-02-26 by lehahoang