ARM-software/SCALE-Sim #24: about OS dataflow definition

jmjmjm123, 2019-04-23:
Hi,thank you very much for the great project!
It really can be a powerful tool for  system architecture exploration.

I have read your paper about  SCALE_sim， but I am confused about the description about OS dataflow.   SHIDIANNAO and  this work:**Co-Design of Deep Neural Nets and Neural Net Accelerators for Embedded Vision Applications  from UC Berkeley**   have different  description about OS dataflow: the PE array map to several out feature map's adjacent activation in the same output channel,  the Cout(output channel ) loop is not unrolled  spatially. 
As for your paper, OS description part says the PE array's  column corresponds to different output channel.
This really confused me and I really am looking forward your reply!

Thanks a lot!

AnandS09, 2019-04-25:
Hi @jmjmjm123 , 
 
Output stationary dataflow refers to the mapping where every processing element is responsible for generating one or more output pixels, with each of them reducing all the required partial sums locally. 

In SCALE-Sim, in case of output stationary dataflow, all PEs in a given column compute the output pixel using the same filter, thus generating pixels for the same output channel.  

Let me know if this helps you.

jmjmjm123, 2019-04-25:
thank your for your reply！
To my understanding， in OS dataflow different PE column should compute the same output channel's 2D feature map.
the image below is from eyeriss.  The OSA  is for CONV layers. PEs map to single output channels , multiple output activations:
[image]

so, I am confused about it.

AnandS09, 2019-04-25:
SCALE-Sim's OS is basically OS-B in the above diagram, where every column is computing pixels for a different m (where m= 1,2, ..., M). In a given column, each row is generating an output pixel e (where e = 1,2, ..., E2) with a given filter (thus in the same output channel). OS-C is a special case of OS-B and is realized when the number of cols in a systolic array is >= number of filters and number of rows is 1.

jmjmjm123, 2019-04-25:
So, OS have different versions and your model use OSB.
I have run mobilenet.csv with  scale.cfg at ws and os. The result show that ws/os have nearly the same performance :
[image]

This work: Co-Design of Deep Neural Nets and Neural Net Accelerators for Embedded Vision Applications from UC Berkeley  which also has  using  simulation to compare ws/os dataflow's performace on mobilenet . Their results show that  OS is 3.3(6.35/1.91) times faster than WS.
[image]

Why the results is so different? Is that because  models are different, or which one is more reasonable？

pakiko24, 2019-04-30:
Hi @jmjmjm123 

I have read your comment and let me try to help you. 

Please @AnandS09  correct me if I am wrong but I think the reason why SCALE-SIM barely see differences in terms of cycles between OS and WS is because the cycle count does not take into account the stalls that might be produced in the pipeline due to the memory accesses. 

Both architectures have different reuse capacities and depending on the exact implementation of the architecture you might reuse more or less data. That is the reason why the paper you comment presents 6x of speedup using WS. Apparently, the implementation they have done for WS, "matches" mobilenets network much better than the OS implementation. This means that their WS implementation reuse more data and therefore reduces both the DRAM and  the SRAM accesses, improving energy efficiency. I can't tell you the exact reason why 6x of speedup is obtained with WS. I would have to read the paper and see  the implementation they have done. (Btw thanks for that reference. It seems interesting!) 

If you want to see what implementation reuse better in SCALE-SIM, you have to analyze the memory accesses file produced by the program. It will give you an idea of the reuse with the SRAM buffers given in the input parameters. Anyway, as I said, you can't see that in terms of cycles, because SCALE-SIM supposes a "perfect memory" when it is calculating that number of cycles. 

In summary and in my opinion, SCALE-SIM is great to see a general insight of what happens within each dataflow such us the cycles computed (supposing a perfect memory) or the percentage of utilization of the PEs. It might also give you an estimation of the energy consumed by the DRAM and the SRAM. However, be careful because SCALE-SIM just gives you a general overview. The real numbers will be different depending on the exact implementation of the architecture. 

I hope to have helped you,
Francisco

AnandS09, 2019-05-18:
Hi @jmjmjm123 / @pakiko24 , 

Thanks guys for bringing this up. This trend is very interesting. 
Also apologies for not reply for long time ( bad time management on my part I guess!) 

Anyhow, @pakiko24 's explanation for SCALE-Sim's OS and WS cycles is correct. However, after going through the accelerator design in the paper you mentioned, it seems like it's not exactly a systolic array. In SCALE-Sim the store-and-forward mechanism is followed everywhere and cycles times are calculated accordingly. 
From what I have understood, the WS and OS data flow they employed is not store and forward based (notice the broadcast bus and input selection in the PE diagram), which might explain the difference in trend. Plus, as @pakiko24 mentioned OS in SCALE-Sim assumes stall free operation, however if stall is taken into account then WS and OS times will be closer to each other assuming similar utilization. 

I need to look deeper into the design of the paper @jmjmjm123 , I will get back if I get more details. 
I hope it makes sense this far. 

Thanks
Anand

HaFred, 2021-04-28:
> Hi @jmjmjm123 / @pakiko24 ,
>
> Thanks guys for bringing this up. This trend is very interesting.
> Also apologies for not reply for long time ( bad time management on my part I guess!)
>
> Anyhow, [...]

Hi Anand, sorry to bring this up. Since it is still open, and it looks interesting to me. I figured might as well put on some thoughts. 
You said that, 
> Plus, as @pakiko24 mentioned OS in SCALE-Sim assumes stall free operation, however if stall is taken into account then WS and OS times will be closer to each other assuming similar utilization.
Is it typo? According to jm123's curve graph, with stall freee operation in OS, the cycles count of WS and OS are already close. If stall is taken, shouldn't it be more different than each other?

closed 2023-05-11 by danh-arm