ARM-software/SCALE-Sim #25: "The syntax of the command is incorrect" error

yuezuegu, 2019-04-25:
When I edit a topology file with Excel on Windows, I am getting the following error: 

> The syntax of the command is incorrect.
> The syntax of the command is incorrect.
> 'mv' is not recognized as an internal or external command,
> operable program or batch file.
> The syntax of the command is incorrect.
> 'mv' is not recognized as an internal or external command,
> operable program or batch file.
> 'mv' is not recognized as an internal or external command,
> operable program or batch file.

The reason, I figured, is that Excel does not put a comma at the end of the line, so the length of elems in run_nets.py becomes less than 9 and it skips the layer even though the line is complete.

AnandS09, 2019-05-18:
@yuezuegu add an extra column in excel or use another editor

yuezuegu, 2019-05-18:
I figured that... I just posted here in case you want to fix it in your code.

closed 2019-05-18 by yuezuegu