ARM-software/SCALE-Sim #26: How to send output activations out of the pipeline.

pakiko24, 2019-04-30:
Hello,

I know that SCALE-SIM does not focus on the exact implementation. However I would like to know if the writes that are performed in SCALE-SIM are realistic. Most of all in the OS architecture. 

Taking into account the current implementation of SCALE-SIM for OS dataflow, it would require one write port for each PE of the array. I mean, each cycle, an entire row of PEs, write into memory. However, currently, it seems as if in different cycles, a different row writes. This would require a huge number of write ports in the output buffer.  Am I wrong? How would be possible to implement this, without incurring in this cost overhead? 

Best regards,
Francisco.

AnandS09, 2019-05-18:
Hi Francisco, 
There are multiple ways of implementing this. One way is what you just describes, in which you are adding multiple ports trading off area and power. Another way is to have a bunch of buffers adjacent to MAC PEs such that you have enough buffers to hold the generated outputs until read out, without taking a stall penalty, here again we are trading off area and power. A simpler yet less performant implementation would be to stall the systolic array until the buffers are empty and ready for next set of partial sums.

Let me know if this makes sense.

Cheers
Anand

pakiko24, 2019-05-23:
Hi Anand,

Those approaches you describe make sense in terms of functionality. And I see it as a good abstraction to evaluate just the dataflow. However, all of them look infeasible in terms of area and power. Have you implemented this on FPGAs or directly on hardware? 
For instance, using the first approach I described would need one port per PE. This means that in an array of 32x32 would be neccesary 1024 ports in SRAM. This looks completely infeasible. Don't you think so?

donghn, 2019-10-17:
@pakiko24  @AnandS09 
Hi. I was confused 1 thing. The number of bytes for each transaction (read or write) from DRAM is set to 10 bytes/cycle. Why some parts of trace file the read or write to DRAM is 14 bytes/cycle, and some other just 3bytes/cycles.
Can I combine it to 1 transaction or device it to many transaction? It is still true to estimate the Power?
thanks so much.

closed 2021-04-30 by whatmough