ARM-software/SCALE-Sim #27: Is there any padding considerations?

zinechant, 2019-05-10:
It seems the code default to 'valid' padding, where resnet employs 'same' padding.

pakiko24, 2019-05-13:
Hi @zinechant , let me try to  answer your question. 

Really, padding does not make sense in SCALE-SIM. The simulator just receives the network architecture through the network file and simulates it using a certain dataflow. Without using the actual values. It just simulates the behaviour of the dataflow based on the network topology. If you want to 'simulate' a padded size, you just have to modify the network architecture file to add as many rows and columns as you wish.

Hope this helps. 

Best regards,
Francisco
email: francisco.munoz2@um.es

AnandS09, 2019-05-18:
Hi @zinechant ,
@pakiko24 is right, SCALE-Sim expects you to provide padded dimensions in the topology .csv file.

zinechant, 2019-06-12:
Thanks. Yes, the explanation makes sense.

zinechant, 2019-06-12:
Sorry, I think there are still issues.

If modifying the input dimension, the number of input values read out from SRAM and DRAM will be changed. I think zero are inserted by any controller rather than read out from SRAM or DRAM in a realistic implementation, 

Of cause, the different is only marginal, but it's not the same.

closed 2021-04-30 by whatmough