ARM-software/SCALE-Sim #28: Run the simulation on computer GPU

manili, 2019-05-26:
Hi all,
I'd like to know if it is technically possible to modify the source code and make the SCALE-Sim run on the computer GPU for faster simulation?
In other words, is it possible to use the computation power of GPUs or other accelerators (like FPGAs) to do the simulation process much faster?

pakiko24, 2019-06-08:
Hi @manili ,

I guess you could use CUDA-python or something like that to modify the code and execute it on GPU. However, I don't think that is obvious and I don't know if it's worth it, since the execution of SCALE-Sim is pretty fast. I have run it many times and I have had no problem at all. Anyway, I think everything is possible and you can code it. And btw, If you do so, it would be interesting to see it :D.

Best regards,
Francisco Muñoz.

whatmough, 2021-04-30:
This is currently out of scope for this project

closed 2021-04-30 by whatmough