ARM-software/SCALE-Sim #29: Bug in transformer configuration file?

zinechant, 2019-06-12:
First I believe the configuration assumes the sentence length is 1 for both encoder and decoder.

I believe the following corresponds to an attention layer (either self or cross attention).
Layer name, IFMAP Height, IFMAP Width, Filter Height, Filter Width, Channels, Num Filter, Strides.
Attn_Xmap,1, 1, 1, 1, 512, 64, 1, 
Attn_Kmap,1, 1, 1, 1, 512, 64, 1, 
Attn_Qmap,1, 1, 1, 1, 512, 64, 1, 
Attn_QxK', 1, 64, 1, 1, 1, 64, 1,
Attn_SFMXxV, 1, 64, 1, 1, 64, 1, 1,

The first 3 corresponding to the mapping and split of the head. No issues there.
'Attn_QxK', 1, 64, 1, 1, 1, 64, 1' is where I think the problem is.
If we follow the rule of conv layer, the output should be a matrix of 64x64. While it should really be a matrix of length x length, and 1x1 in this case since the length is 1.

aqeelanwar, 2019-06-26:
Hi Zinechant

I have had the same confusion and few other confusion but wasnt able to verify. Here is what I think it should have been. 

[image]

1. Sentence length is 1
2. The embedding layer basically multiplies the input one-hot (1x1x33708) vector with a learned embedding-matrix to generate the 1x1x512 vector. Its like a dense layer except there is no bias. Hence the channel is equal to the dimension of input vector, and Num filter equals the output dimension.

2. Attn_QxK' should be a scalar . i.e. multiplication of 1x64 dimensional vector with 64x1 dimensional vector. Hence  Its again like a dense layer with input 1x1x64 and output 1x1x1. Here the input feature map is the vector Q and the weight being multiplied with is vector K'

3. Softmax(QK')*V takes input the scalar and multiples it with the vector V. Its dense formulation will be going from input 1x1x1 to 1x1x64. QK' can be taken as the input feature map and the weights to be V. This dense layer will then generate the output in the required way

4. The concatenation step shouldnt be there after every head calculation. Rather once all the heads have been calculated, we need to carry out a multiplication between the concatenated output of all heads (1x512) with a matrix that generates the output 1x64. Again dense layer representaion.

I am not sure if this approach is right. What do you think?

whatmough, 2021-04-30:
We'll take this discussion over to v2, here: https://github.com/scalesim-project/scale-sim-v2

closed 2021-04-30 by whatmough