ARM-software/SCALE-Sim #37: Number of bytes for each Read or Write transaction to DRAM.

donghn, 2019-10-17:
Hi. I was confused 1 thing. The number of bytes for each transaction (read or write) from DRAM is set to 10 bytes/cycle. Why some parts of trace file the read or write to DRAM is 14 bytes/cycle, and some other just 3bytes/cycles.
Can I combine it to 1 transaction or device it to many transaction? It is still true to estimate the Power?
thanks so much.

donghn, 2019-10-17:
[image]
Can we change words_per_cycle to: 
'''
 words_per_cycle  = default_read_bw
'''

AnandS09, 2019-10-18:
Hi @donghn 

>The number of bytes for each transaction (read or write) from DRAM is set to 10 bytes/cycle.
This is not fixed, it changes based on the read demand for no-stall operation.

>Can I combine it to 1 transaction or device it to many transaction?
>For example, we transfer 3000 bytes in 1000 cycles. Instead of transference 3 bytes/cycle Can I transfer 6 bytes every 2 cycles in only 1 cycle and leave idle in another cycle?
This is not a configurable parameter in SCALE-sim, currently, the tool would spread out the transaction such that BW requirement is not exaggerated.

donghn, 2019-10-20:
@AnandS09 Thank so much for your reply.
So, you mean read/write to DRAM, do not depend on datawidth of DRAM? (just depend on the demand of operation)? is it true in reality?
As I see in the code. After generate the SRAM trace file. We trace the DRAM write/read from that file. We go through all transactions of SRAM, Whenever the SRAM is full we start to trace the DRAM read/write depending on how many clocks passed and how many bits transferred. So, every clocks we transfer [num_clocks/num_bytes] bytes to DRAM.
am I right? 
If that is true. Can I custom for my DRAM which can read/write 8bytes every cycle?
If not. There is any way to do that?
thank you.

whatmough, 2021-04-30:
This is basically a simplifying assumption that we use for the simulator.  In practice there is of course a dependence on the DRAM/bus bandwidth.

closed 2021-04-30 by whatmough