ARM-software/SCALE-Sim #40: Different stride on each Conv2D dimension

Kuronogard, 2020-08-06:
Great work on the simulator.
I'm trying to simulate a network that has a different stride on each dimension, but the simulator only accepts a "strides" parameters, which is applied to both h and w dimensions of the ifmap... why is that? Is there anything preventing the implementation of separated h and w strides?

I started reading the code, but I am not comfortable modifying the formulas, since there are many undocumented details that I still don't fully understand...

Thank you.

AnandS09, 2020-08-06:
@Kuronogard: Thanks for bringing this into notice. I will add a patch for this shortly.

Kuronogard, 2020-08-06:
Thank you, @AnandS09, that will be very helpfull!

Kuronogard, 2020-08-11:
I submitted a pull-request  (#41) solving this.
I checked the results for the three dataflows with different stride configurations and everything seems to be correct. Besides that, I found and fixed what seemed to be a bug... When in IS, the filter SRAM addresses keep increasing when they should restart to the beginning of the filters. 

Please review the commits and merge them if appropriate =)

whatmough, 2020-08-11:
Many thanks for your hard work Dennis!!

AnandS09, 2020-08-11:
Thank you @Kuronogard I will review and take the patch in. 
Appreciate the addition and the fix. 

+ If you can send a description of the bug in IS implementation, it would help me with the review and issue documentation. 

Thanks
Anand

Kuronogard, 2020-08-12:
Yes, of course.

**config file:**
[general]
run_name = "TEST_CONFIG_15x15_is"

[architecture_presets]
ArrayHeight:    15
ArrayWidth:     15
IfmapSramSz:    512
FilterSramSz:   512
OfmapSramSz:    256
IfmapOffset:    0
FilterOffset:   10000000
OfmapOffset:    20000000
Dataflow:       is

**Topology file:**
Layer name, IFMAP Height, IFMAP Width, Filter Height, Filter Width, Channels, Num Filter, Strides,
Conv, 80, 20, 2, 3, 10, 5, 1,

As I understand, the ifmaps should be loaded into the array (from above), and the filters fed from the left.
The expected sram read pattern is:
1 - read 15x15 ifmaps
2 - read first 15 weights from each filter
3 - read following 15x15 ifmaps
4 - read following 15x15 weights from each filter
5 - continue until the inputs and filters are exhausted for those ofmaps.
6 - read 15x15 ifmaps corresponding to the next conv window
7 - read first 15 weights from each filter
8 - ...

However, in the step 7 (line 116 from __Conv_sram_read.csv_ file), the filter addresses increase instead of returning to 0.

The problem was that the list of filter addresses (_all_filt_addr_list_ from _sram_traffic_is.py#63_) was initialized just once before the h_fold and v_fold loops (inside which, said list is updated). The solution was to reset _all_filt_addr_list_ just before the v_fold loop.

whatmough, 2021-04-30:
PR taken

closed 2021-04-30 by whatmough