ARM-software/SCALE-Sim #42: Different NoC topologies simulation with SCALE-Sim?

manili, 2020-08-31:
Hi everyone.
As far as I know different NoC topologies could affect the overall performance of the accelerators due to the congestion of packets between PEs/GB. As an example the Microswitch is causing much less latency compared to the Mesh topology so it will use much less clock cycles in a real world accelerator [https://doi.org/10.1145/3130218.3130230].
So I'm really wondering to know if it is possible to simulate different NoC topologies with SCALE-Sim?

Best regards,
Manili

AnandS09, 2020-09-01:
Hi Manili, 

The point you raised is relevant in larger context of spatial accelerators. However, please note that SCALE-Sim's internal accelerator is strictly a systolic array. 
Within a systolic array the data movement is highly deterministic, following a store-and-forward pattern. 
Given that there are no stalls introduced while feeding in the data from the memory, the data movement never stops. 

Since, the interconnection is fixed in a systolic array based system no additional NoC is supported in SCALE-Sim.

Thanks
Anand

manili, 2020-09-01:
Hello Anand,
Thanks a lot for the reply. Excellent point, I'm sorry I didn't notice that.

Best regards,
Manili

closed 2020-09-01 by manili