ARM-software/SCALE-Sim #43: Is it possible to simulate mixed-precision architecture like Bit Fusion

sunlex0717, 2020-10-21:
Hi:

I am wondering if it is technically possible to use the simulator to simulate Bit Fusion architecture(Bit Fusion: Bit-Level Dynamically Composable Architecture for Accelerating Deep Neural Networks) which supports mixed-precision computation(2bit, 4bit, 8bit and 16bit)?

Kind Regards

AnandS09, 2020-10-21:
Hi @sunlex0717,

You raise an interesting point. SCALE-Sim at this point specifically targets systolic arrays. 
I am unsure if bit-fusion architecture can be modeled as such. 
If you think otherwise, please let me know. 

Thanks
Anand

sunlex0717, 2020-10-28:
Hi Anand:
Thanks for your reply.
BitFusion is also a systolic-array-like architecture with some extra specialized control logics to support  flexible bit operations. 
I checked SCALE-Sim briefly. It seems that users can only change the size (e.g. array size, sram size and dram size ) and dataflow strategy(OS,WS,IS).
Is SCALE-Sim able to simulate this kind of property(e.g. customized control path beside standard systolic-array architecture)? 
Kind Regards,
Wei

AnandS09, 2020-12-17:
Hi @sunlex0717 , 

Sorry for the late reply. At the moment, custom control paths are not supported. 

-Anand

Also please close this issue, as it seems more like a future feature recommendation, instead of a bug in the tool. Thanks!

AnandS09, 2021-01-08:
Hi @whatmough ,

I think we can close this issue. 

Thanks,
Anand

closed 2021-01-11 by whatmough