ARM-software/SCALE-Sim #44: Energy Calculations

05Tushar, 2020-12-17:
Hi!
In the paper 'SCALE-Sim: Systolic CNN Accelerator Simulator', a lot of discussions are based on energy. But Scale-Sim doesn't inherently calculate energy. So how was energy calculated?

Thanks!

AnandS09, 2020-12-17:
Hi @05Tushar ,

The energy numbers in the SCALE-Sim paper are calculated by multiplying the access counts to the memory and the MAC activity with the energy consumed per access/ energy per mac from our RTL implementations. 
Unfortunately, we cannot release the RTL numbers. 
Hope this helps. 

-Anand

Closing this as it is not an issue with the tool.

05Tushar, 2020-12-18:
Thankyou!

05Tushar, 2020-12-22:
Sorry to reopen it again. I just need a quick clarification on your comment above. 
You said energy/access or mac was obtained using RTL simulations. Were access counts like inter PE transfer, register file accesses, etc were also obtained from RTL simulations or were they calculated using Scale Sim only.
Thanks 
Tushar Bansal

AnandS09, 2020-12-22:
>Were access counts like inter PE transfer, register file accesses, etc were also obtained from RTL simulations or were they calculated using Scale Sim only.

SCALE-Sim provides access counts, all the energy numbers are obtained from RTL synthesis. There is no register file in systolic array, but the costs for peer to peer communication is captured in synth.

closed 2020-12-29 by 05Tushar