ARM-software/SCALE-Sim #46: Bug in generating ifmap trace for ws

diwu1990, 2020-12-30:
In this line, https://github.com/ARM-software/SCALE-Sim/blob/5825f6cee26bac223ce6019fcfc4f384c1142cb6/sram_traffic_ws.py#L382
the shape is not formatted to int, leading to new row is never generated here https://github.com/ARM-software/SCALE-Sim/blob/5825f6cee26bac223ce6019fcfc4f384c1142cb6/sram_traffic_ws.py#L441

Hope this bug can be fixed soon, so that others wont come across this problem.

BTW, I am not sure whether somewhere else have this problem.

AnandS09, 2020-12-31:
Thanks, @diwu1990, 

If you have already implemented it, can you please send a pull request. 
Also please include the inputs you got to catch this bug, it would help me test the solutions. 

Thanks
Anand

AnandS09, 2021-01-08:
Thanks, @diwu1990, 

For sending the fix, I appreciate you taking your time contributing to the project.
I am still in the process of testing the patch and would take it in once my tests pass.

Cheers,
Anand

AnandS09, 2021-04-22:
Hi @whatmough : 

I think we have already integrated it. We can close this issue. 

Thanks,
Anand

closed 2021-04-30 by whatmough