ARM-software/SCALE-Sim #49: A possible bug in cycle computations in gen_trace_ofmap function of sram_traffic_is.py

sparsh0mittal, 2021-01-20:
See https://github.com/ARM-software/SCALE-Sim/blob/master/sram_traffic_is.py

In  gen_trace_ofmap function, currently, "local_cycle += 1" happens on line 376, which is outside the for loop, "for p in range(parallel_window)"

I think that local_cycle should be incremented inside this for loop.
A sample sram_write_trace_file with existing code is attached [file_Conv5_sram_write.xlsx](https://github.com/ARM-software/SCALE-Sim/files/5841929/file_Conv5_sram_write.xlsx), where I simulated
Layer name, IFMAP Height, IFMAP Width, Filter Height, Filter Width, Channels, Num Filter, Strides
Conv5,32,32,3,3,1,1,1
Array-size was 108*108

Value of parallel_window was 9, so as you can see from the attached file, in the same cycle, say 90, it is writing  9 values: 
90,20000000, , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,20000108, , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,20000216, , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,20000324, , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,20000432, , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,20000540, , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,20000648, , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,20000756, , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,20000864, , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,

Since there are only 108 columns, this seems to be possible.

AnandS09, 2021-04-22:
Hi @sparsh0mittal : 

Thanks for bringing this issue to notice. Can you please attach the topology csv file and the config file so that I can reproduce this issue? 

Thanks 
Anand

sparsh0mittal, 2021-04-26:
The topology file is just this:
Layer name, IFMAP Height, IFMAP Width, Filter Height, Filter Width, Channels, Num Filter, Strides
Conv5,32,32,3,3,1,1,1

In config file, array-size is 108*108 and dataflow is input-stationary. Here it is:
[general]
run_name = ""

[architecture_presets]
ArrayHeight:    108
ArrayWidth:     108
IfmapSramSz:    512
FilterSramSz:   512
OfmapSramSz:    256
IfmapOffset:    0
FilterOffset:   10000000
OfmapOffset:    20000000
Dataflow:       is

closed 2023-05-11 by danh-arm