ARM-software/SCALE-Sim #50: Using word size > 1B in the tool has no impact on DRAM bytes read/written

agarwal-ayushi, 2021-02-07:
Hello, 
I am trying to use the framework for word size > 1B (which is the default right now). 
I did the necessary changes in the code for setting the word size to 4B in my case. 

In the [dram_trace.py](https://github.com/ARM-software/SCALE-Sim/blob/5825f6cee26bac223ce6019fcfc4f384c1142cb6/dram_trace.py#L55), we print the dram traces according to the number of words per cycle (and so we see consecutive addresses in the trace even though they are multi-byte words). 
However, in the [trace_gen_wrapper.py](https://github.com/ARM-software/SCALE-Sim/blob/5825f6cee26bac223ce6019fcfc4f384c1142cb6/trace_gen_wrapper.py#L192), now when we read the trace to generate the detailed log, we just count the number of addresses in the trace and print them as the number of DRAM bytes. In this case, isn't that supposed to be multiplied by the word size to determine the actual number of bytes read from the DRAM?

Is there a gap in my understanding? If yes, please guide with the changes for using multi-byte words. 

-Ayushi

AnandS09, 2021-04-22:
@agarwal-ayushi : WORD_SIZE > 1 is not currently supported in this version. 

Please note that new enhancements will be reflected in the v2 released a few days ago [https://github.com/scalesim-project/scale-sim-v2]

We will support this repository for bug-fixes only. 

Thanks
Anand

AnandS09, 2021-04-22:
@whatmough : Hi Paul, we can close this issue.

closed 2021-04-30 by whatmough