ARM-software/SCALE-Sim #51: gemm and gemv simulation

ramcn, 2021-02-23:
Hello,

I am trying to measure the runtime and utilization of a GEMM (M=256, K=768 and N=256) and GEMV (M=1, K=768 and N=256) operation on a TPU of size 128x128 in weight stationary dataflow setting.

Below is the parameter setting, output cycles and utilization:
CONFIG | IF HEIGHT | IF WIDTH | FILTER HEIGHT | FILTER WIDTH (SC) | CHANNELS | NUM FILTER | STRIDE | CYCLES | UTILIZATION
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
GEMM | 256 | 768 | 768 | 1 | 1 | 256 | 1 | 4608 | 100
GEMV | 1 | 768 | 768 | 1 | 1 | 256 | 1 | 4608 | 100

It will be great if I can get some insights about the below questions.
1. Is this the right parameter settings to simulate the GEMM and GEMV of the above mentioned size.
2. The 4608 cycles taken by GEMM represents 128x12 cycles to load the filter and 256x12 cycles for the systolic computation. Is this understanding correct?
3. Why does GEMV show the same cycles and utilization as GEMM. I was expecting it to be much lesser.

Thank you,
Ram

ramcn, 2021-03-02:
I found the below description of mapping scalesim parameters to GEMM from the scalesim paper:

"For multiplying matrices of size M × K and K × N , the dimension M is mapped to SR(IF height), dimension N is mapped to SC (num filters) and the dimension K to T (filter height)". 

So specifying IF width seemed redundant.  And I repeated the experiment with the below updated configuration for a GEMM (M=256, K=768 and N=256) and GEMV (M=1, K=768 and N=256)

CONFIG | IF HEIGHT  (SR, M) | IF WIDTH | FILTER HEIGHT  (T, K) | FILTER WIDTH (SC) | CHANNELS | NUM FILTER (SC, N) | STRIDE | CYCLES | UTILIZATION
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
GEMM | 256 | 1 | 768 | 1 | 1 | 256 | 1 | 4608 | 100
GEMV | 768 | 1 | 1 | 1 | 1 | 256 | 1 | 899 | 1.56
GEMV | 1 | 1 | 768 | 1 | 1 | 256 | 1 | 4608 | 100

The utilization and cycles for GEMM match with the analytical model. (load time = 1536 and streaming time = 3072 total 4608 cycles). But I haven't been able to map the simulation of the 2 GEMV configuration to the analytical model. (expected load time = 1536 and stream time = 768 and total 2304 cycles).

AnandS09, 2021-04-22:
Hi @ramcn : 

Not sure if I understand the issue correctly. 
As far as I know the loading and streaming times for any operation will depend on the dataflow choice along with the parameters you mentioned. 

Can you give me more details about what dataflow and workload parameters you are using? 
Please attach the topology files and config file for your runs. 

Thanks
Anand

closed 2023-05-11 by danh-arm