ARM-software/SCALE-Sim #52: How to feed the DRAM read/write bandwidth into DRAMSim2 for power/area estimation?

HaFred, 2021-03-29:
Hi @AnandS09 , I am trying to integrate the SCALE-Sim result with DRAMSim2. But I noticed that 
* The output traces of SCALE-Sim is not matched with the required DRAMSim2 traces, which is more like instruction traces generated by gem5?
* The SCALE-Sim paper mentions that using read/write bandwidth as input of DRAMSim2 simulation. However, I did not find out how to do it (referring to https://github.com/umd-memsys/DRAMSim2)
* There is also an unsolved related issue https://github.com/ARM-software/SCALE-Sim/issues/35

Thank you very much.

AnandS09, 2021-04-22:
Hi @HaFred : 

Thanks for raising the issue. In the paper @yuhao , wrote a custom script, which can parse the DRAM traces, and bandwidth requests generated by SCALE-Sim and hence generate DRAMSim parameters and requests. 
Since DRAM is not an inherent part of a systolic array like accelerator we did not include it into SCALE-Sim and there are no plans to include it in future as well. 

I hope this helps to clarify your questions.
Closing this issue since it does not report and bug/enhancement to the tool. 

Thanks
Anand

AnandS09, 2021-04-22:
@whatmough : Hi Paul, please close this issue.

HaFred, 2021-04-23:
Got it. Thank you.

closed 2021-04-23 by HaFred