ARM-software/SCALE-Sim #54: about DRAM, SRAM

gksruf, 2021-06-08:
Hi, thanks for your great works.

I want to ask about SRAM size. is the unit of SRAM size on configuration file KB?
and is there any way to set DRAM maximum bandwidth?

AnandS09, 2021-06-08:
Hi @gksruf , 

Thank for the encouraging words. Please find the answer your questions below. 

>I want to ask about SRAM size. is the unit of SRAM size on configuration file KB?
That is correct

>is there any way to set DRAM maximum bandwidth?
Yes, but not in this version of the tool. You can set the interface bandwidth in the v2 of our simulator. (https://github.com/scalesim-project/scale-sim-v2)

>Does power estimation include NoC, MAC and Register?
This does does not perform power estimation natively. However, you can use the activity counts generated by the tool to estimate energy and power for the system modeled.

>I don't understand what (DRAM_IFMAP_start, DRAM_IFMAP_stop, DRAM_IFMAP_bytes) mean.
DRAM_IFMAP_start: The cycle at which elements of IFMAP matrix starts to be fetched from DRAM (-ve cycles indicate that the transactions are done before the compute starts)
DRAM_IFMAP_stop: Indicates when DRAM transactions stop
DRAM_IFMAP_bytes: Number of bytes transacted

>alexnet_Conv5_sram_write.csv also. (The output has only number)
Its the cycle accurate trace alexnet conv 5, for outputs generated from array to SRAM. 
The trace format is `cycle, addr1, addr2, ... , addrC` (C = num cols)

>Assuming that we use 8-bit machine, (1024x1024/65567)= 16bytes/clyce are required only for weights.
Am i wrong with it?
Need more info on sizes of the SRAM. However, compute cycles do not account for the cycles taken to bring data into the SRAM before the compute started, which might be why you see a lower predicted BW.
Plus the DRAM accesses for IFMAP, Filter, and OFMAP happen independently in parallel.

gksruf, 2021-06-09:
Thanks for your clear answer.

But i don't still figure out my last question.
is The Cycles for compute not total time to complete process? (you mean it does not include DRAM in, out, fill buffer and drainout buffer ) 
If it is not, how can i know the total time of process ?

Thanks.

AnandS09, 2021-06-13:
>If it is not, how can i know the total time of process ?
The total time is not explicitly reported by this version v1. In this version, the objective is to focus more on how the systolic array dimensions and mapping affect the execution of a workload. 
The `details.csv` file captures the time when the memory operations start, so you parse this file to calculate the total time taken for any layer.

AnandS09, 2021-06-13:
@whatmough : We can mark this issue as thread and close this if there are no followup questions 
@gksruf : Please close the issue if you think all your queries are answered. 

Thanks
Anand

closed 2021-06-14 by gksruf